anujpathania/HotSniper #1: Could not locate files when creating floorplan and thermal model based on McPAT area estimations using thermallib

jinweizhang001, 2021-01-11:
Hi, firstly thank you for making this tool chain open source. I am trying to run the simulation for a 2x2 core configuration instead of the 8x8 manycore that is given in the demo. Following the configuration checklist, I first set the `NUMBER_CORES = 64` in file `simulationcontrol/config.py` to be `NUMBER_CORES = 4`. Then I set `total_cores = 4` in the file `config/base.cfg`. 

However, when I try to create floorplan and corresponding thermal model as the Configuration Checklist shows, I could not locate the required file `thermallib`, neither the `create_thermal_model.py`. Are these files included in this tool package or to be downloaded from another location? Thank you very much for your help.

ma-rapp, 2021-01-11:
Hi, thank you for pointing out the missing part.

There are two options on how to set up the thermal model in HotSniper:
1) use your own floorplan and use MatEx to create a thermal model from it (often called eigendata.bin).
2) use the thermallib to create a simple floorplan and corresponding thermal model.

I updated the README to describe this process better. I also uploaded the missing thermallib to github.

Please let me know if you experience any further trouble and I will assist you.

jinweizhang001, 2021-01-12:
> Hi, thank you for pointing out the missing part.
>
> There are two options on how to set up the thermal model in HotSniper:
>
>     1. use your own floorplan and use MatEx to create a thermal model from [...]

Thanks very much, it solves this problem finally. I actually modified the code `from thermallib.chip import Chip` in file `thermallib/thermallibtest.by` with `from chip import Chip`. I think it is a typo but this is easy to correct.

closed 2021-01-12 by jinweizhang001