anujpathania/HotSniper #2: Step "4- Running the Simulations" doesn't work.

Roozbeh2429, 2021-01-29:
It seems it doesn't have permission to some files.

anujpathania, 2021-01-30:
Hi, Please add more details (and all the logs you may have) to your issue for us to investigate and assist.

Roozbeh2429, 2021-01-30:
Hi, thank you, I'm using Ubuntu 20.04.1 LTS,
here is part of the logs for command: "PYTHONIOENCODING="UTF-8" python3 run.py"

IOError: [Errno 2] No such file or directory: '/home/roozbeh/projects/HotSniper/HotSniper/benchmarks/energystats-temp.py'
257664881  [  206]  [ 0]  [py_stats.cc         : 152]  *WARNING* Stats callback: return value must be (convertable into) 64-bit unsigned integer
sh: 1: /home/roozbeh/projects/HotSniper/HotSniper/tools/mcpat.py: Permission denied
Traceback (most recent call last):
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 103, in get_stat
    self.update()
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 90, in update
    power = self.run_power(self.name_last, current)
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 160, in run_power
    execfile(outputbase + '.py', {}, result)
IOError: [Errno 2] No such file or directory: '/home/roozbeh/projects/HotSniper/HotSniper/benchmarks/energystats-temp.py'
257684405  [  206]  [ 0]  [py_stats.cc         : 152]  *WARNING* Stats callback: return value must be (convertable into) 64-bit unsigned integer
sh: 1: /home/roozbeh/projects/HotSniper/HotSniper/tools/mcpat.py: Permission denied
Traceback (most recent call last):
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 103, in get_stat
    self.update()
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 90, in update
    power = self.run_power(self.name_last, current)
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 160, in run_power
    execfile(outputbase + '.py', {}, result)
IOError: [Errno 2] No such file or directory: '/home/roozbeh/projects/HotSniper/HotSniper/benchmarks/energystats-temp.py'
257703375  [  206]  [ 0]  [py_stats.cc         : 152]  *WARNING* Stats callback: return value must be (convertable into) 64-bit unsigned integer

[Scheduler]: Time 42.000.000 ns [Active Tasks =  1 | Completed Tasks = 0 | Queued Tasks = 0 | Non-Queued Tasks  = 0 | Free Cores = 48 | Active Tasks Requirements = 16 ] 

[Scheduler]: DVFS Control Loop invoked at 42.000.000 ns
[Scheduler][DVFS_MAX_FREQ]: Core  0: P=-1.000 W f=1000 MHz T=-1.0 °C utilization=1.000
[app0] [HOOKS] Total time spent in ROI: 0.013s
[TRACE:0] -- DONE --

[Scheduler]: Thread 0 from Task 0 Exiting at Time 42.349.253 ns

[Scheduler]: Releasing Core 0 from Thread 0

[Scheduler]: Task 0 Finished.

[Scheduler]: Releasing Core 0 from Task 0

[Scheduler]: Releasing Core 1 from Task 0

[Scheduler]: Releasing Core 2 from Task 0

[Scheduler]: Releasing Core 3 from Task 0

[Scheduler]: Releasing Core 4 from Task 0

[Scheduler]: Releasing Core 5 from Task 0

[Scheduler]: Releasing Core 6 from Task 0

[Scheduler]: Releasing Core 7 from Task 0

[Scheduler]: Releasing Core 8 from Task 0

[Scheduler]: Releasing Core 9 from Task 0

[Scheduler]: Releasing Core 10 from Task 0

[Scheduler]: Releasing Core 11 from Task 0

[Scheduler]: Releasing Core 12 from Task 0

[Scheduler]: Releasing Core 13 from Task 0

[Scheduler]: Releasing Core 14 from Task 0

[Scheduler]: Releasing Core 15 from Task 0
[... 29 lines cut ...]
  File "/home/roozbeh/projects/HotSniper/HotSniper/tools/cpistack_data.py", line 9, in __init__
    data_raw = sniper_lib.get_results(jobid = jobid, resultsdir = resultsdir, config = config, stats = stats, partial = partial)
  File "/home/roozbeh/projects/HotSniper/HotSniper/tools/sniper_lib.py", line 47, in get_results
    results = parse_results_from_dir(resultsdir, partial = partial, metrics = metrics)
  File "/home/roozbeh/projects/HotSniper/HotSniper/tools/sniper_lib.py", line 179, in parse_results_from_dir
    results += stats.parse_stats((k1, k2), ncores, metrics = metrics)
  File "/home/roozbeh/projects/HotSniper/HotSniper/tools/sniper_stats.py", line 8, in parse_stats
    v2 = self.read_snapshot(k2, metrics = metrics)
  File "/home/roozbeh/projects/HotSniper/HotSniper/tools/sniper_stats_sqlite.py", line 44, in read_snapshot
    raise ValueError('Invalid prefix %s' % prefix)
ValueError: Invalid prefix roi-end
Traceback (most recent call last):
  File "run.py", line 248, in <module>
    main()
  File "run.py", line 243, in main
    example()
  File "run.py", line 235, in example
    run(['{:.1f}GHz'.format(freq), 'maxFreq', 'slowDVFS'], get_instance(benchmark, parallelism, input_set='simsmall'))
  File "run.py", line 102, in run
    cpistack = subprocess.check_output(['python', os.path.join(SNIPER_BASE, 'tools/cpistack.py')], cwd=BENCHMARKS)
  File "/usr/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['python', '/home/roozbeh/projects/HotSniper/HotSniper/tools/cpistack.py']' returned non-zero exit status 1

ma-rapp, 2021-01-31:
Hi, it seems you have pasted the end of the simulation output. Sometimes an error ocurrs as a consequence of another error. The original error might be a different one.

Please paste the beginning of the simulation output until the first error appears.

Roozbeh2429, 2021-01-31:
Hi. thank you. you are right. here is the first part:

root@364426c6e93b:/home/roozbeh/projects/HotSniper/HotSniper/simulationcontrol# PYTHONIOENCODING="UTF-8" python3 run.py
running parsec-blackscholes-simsmall-15 with configuration 1.0GHz+maxFreq+slowDVFS
-n 64 -c gainestown --benchmarks=parsec-blackscholes-simsmall-15 --no-roi --sim-end=last -senergystats:1000000 -speriodic-power:1000000
[app0] [PARSEC] Benchmarks to run: blackscholes
[app0] 
[app0] [PARSEC] [========== Running benchmark blackscholes ==========]
[app0] [PARSEC] Setting up run directory.
[app0] [PARSEC] Unpacking benchmark input 'simsmall'.
[app0] in_4K.txt
[app0] [PARSEC] Running '/home/roozbeh/projects/HotSniper/HotSniper/record-trace --routine-tracing -o /tmp/tmptR6mo0/run_benchmarks -e 1 -r 1 -s 0 -- /home/roozbeh/projects/HotSniper/HotSniper/benchmarks/parsec/parsec-2.1/pkgs/apps/blackscholes/inst/amd64-linux.gcc-sniper/bin/blackscholes 15 in_4K.txt prices.txt':
[app0] [PARSEC] [---------- Beginning of output ----------]
[SNIPER] Start
[Scheduler]: Setting Arrival Time for Task 0 (parsec-blackscholes-simsmall-15) to 0 ns
[Scheduler] [Info]: Initializing mapping policy
[Scheduler] [Info]: Initializing DVFS policy
Executing Python script /home/roozbeh/projects/HotSniper/HotSniper/benchmarks/sim.scripts.py

[Scheduler]: Trying to map Thread  0 from Task 0 at Time 0 ns

[Scheduler]: Trying to schedule Task 0 at Time 0 ns

[Scheduler]: Task 0 put into execution queue. 
[Scheduler]: Assigning Core 0 to Task 0
[Scheduler]: Assigning Core 1 to Task 0
[Scheduler]: Assigning Core 2 to Task 0
[Scheduler]: Assigning Core 3 to Task 0
[Scheduler]: Assigning Core 4 to Task 0
[Scheduler]: Assigning Core 5 to Task 0
[Scheduler]: Assigning Core 6 to Task 0
[Scheduler]: Assigning Core 7 to Task 0
[Scheduler]: Assigning Core 8 to Task 0
[Scheduler]: Assigning Core 9 to Task 0
[Scheduler]: Assigning Core 10 to Task 0
[Scheduler]: Assigning Core 11 to Task 0
[Scheduler]: Assigning Core 12 to Task 0
[Scheduler]: Assigning Core 13 to Task 0
[Scheduler]: Assigning Core 14 to Task 0
[Scheduler]: Assigning Core 15 to Task 0

[Scheduler]: Setting Affinity for Thread 0 from Task 0 to Core 0

[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Sniper using SIFT/trace-driven frontend
[SNIPER] Running full application in DETAILED mode
[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Enabling performance models
[SNIPER] Setting instrumentation mode to DETAILED
[app0] PARSEC Benchmark Suite Version 2.1
[app0] [HOOKS] PARSEC Hooks Version 1.2

[Scheduler]: Time 1.000.000 ns [Active Tasks =  1 | Completed Tasks = 0 | Queued Tasks = 0 | Non-Queued Tasks  = 0 | Free Cores = 48 | Active Tasks Requirements = 16 ] 

[Scheduler]: DVFS Control Loop invoked at 1.000.000 ns
[Scheduler][DVFS_MAX_FREQ]: Core  0: P=-1.000 W f=4000 MHz T=-1.0 °C utilization=1.000
[SNIPER] Setting frequency for core 0 in DVFS domain 0 to 1000 MHz
[SNIPER] Setting frequency for core 1 in DVFS domain 1 to 1000 MHz
[SNIPER] Setting frequency for core 2 in DVFS domain 2 to 1000 MHz
[SNIPER] Setting frequency for core 3 in DVFS domain 3 to 1000 MHz
[SNIPER] Setting frequency for core 4 in DVFS domain 4 to 1000 MHz
[SNIPER] Setting frequency for core 5 in DVFS domain 5 to 1000 MHz
[SNIPER] Setting frequency for core 6 in DVFS domain 6 to 1000 MHz
[SNIPER] Setting frequency for core 7 in DVFS domain 7 to 1000 MHz
[SNIPER] Setting frequency for core 8 in DVFS domain 8 to 1000 MHz
[SNIPER] Setting frequency for core 9 in DVFS domain 9 to 1000 MHz
[SNIPER] Setting frequency for core 10 in DVFS domain 10 to 1000 MHz
[SNIPER] Setting frequency for core 11 in DVFS domain 11 to 1000 MHz
[SNIPER] Setting frequency for core 12 in DVFS domain 12 to 1000 MHz
[SNIPER] Setting frequency for core 13 in DVFS domain 13 to 1000 MHz
[... 676 lines cut ...]
    power = self.run_power(self.name_last, current)
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 160, in run_power
    execfile(outputbase + '.py', {}, result)
IOError: [Errno 2] No such file or directory: '/home/roozbeh/projects/HotSniper/HotSniper/benchmarks/energystats-temp.py'
1252463    [ 5780]  [ 0]  [py_stats.cc         : 152]  *WARNING* Stats callback: return value must be (convertable into) 64-bit unsigned integer
sh: 1: /home/roozbeh/projects/HotSniper/HotSniper/tools/mcpat.py: Permission denied
Traceback (most recent call last):
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 103, in get_stat
    self.update()
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 90, in update
    power = self.run_power(self.name_last, current)
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 160, in run_power
    execfile(outputbase + '.py', {}, result)
IOError: [Errno 2] No such file or directory: '/home/roozbeh/projects/HotSniper/HotSniper/benchmarks/energystats-temp.py'
1274246    [ 5780]  [ 0]  [py_stats.cc         : 152]  *WARNING* Stats callback: return value must be (convertable into) 64-bit unsigned integer
sh: 1: /home/roozbeh/projects/HotSniper/HotSniper/tools/mcpat.py: Permission denied
Traceback (most recent call last):
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 103, in get_stat
    self.update()
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 90, in update
    power = self.run_power(self.name_last, current)
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 160, in run_power
    execfile(outputbase + '.py', {}, result)
IOError: [Errno 2] No such file or directory: '/home/roozbeh/projects/HotSniper/HotSniper/benchmarks/energystats-temp.py'
1293682    [ 5780]  [ 0]  [py_stats.cc         : 152]  *WARNING* Stats callback: return value must be (convertable into) 64-bit unsigned integer

ma-rapp, 2021-02-01:
This is the root error:
``sh: 1: /home/roozbeh/projects/HotSniper/HotSniper/tools/mcpat.py: Permission denied``

The following looks suspicious to me:
``root@[container-id]:/home/[...]/simulationcontrol#``
You are using root. On my system, it looks like this:
``I have no name!@[container-id]:/home/[...]/simulationcontrol$``

Did you modify the ``docker/Makefile`` or any other file to be root? Do you use ``sudo``?

Roozbeh2429, 2021-02-01:
Thank you for your answer. I changed the docker to the non-root user and I did all the steps with the non-root user. but steel the same error on step 4:

I have no name!@0ce67a0de19e:/home/roozbeh/projects/HotSniper/HotSniper/simulationcontrol$ PYTHONIOENCODING="UTF-8" python3 run.py
running parsec-blackscholes-simsmall-15 with configuration 1.0GHz+maxFreq+slowDVFS
-n 64 -c gainestown --benchmarks=parsec-blackscholes-simsmall-15 --no-roi --sim-end=last -senergystats:1000000 -speriodic-power:1000000
[app0] [PARSEC] Benchmarks to run: blackscholes
[app0] 
[app0] [PARSEC] [========== Running benchmark blackscholes ==========]
[app0] [PARSEC] Setting up run directory.
[app0] [PARSEC] Unpacking benchmark input 'simsmall'.
[app0] in_4K.txt
[app0] [PARSEC] Running '/home/roozbeh/projects/HotSniper/HotSniper/record-trace --routine-tracing -o /tmp/tmpnzbb6z/run_benchmarks -e 1 -r 1 -s 0 -- /home/roozbeh/projects/HotSniper/HotSniper/benchmarks/parsec/parsec-2.1/pkgs/apps/blackscholes/inst/amd64-linux.gcc-sniper/bin/blackscholes 15 in_4K.txt prices.txt':
[app0] [PARSEC] [---------- Beginning of output ----------]
[SNIPER] Start
[Scheduler]: Setting Arrival Time for Task 0 (parsec-blackscholes-simsmall-15) to 0 ns
[Scheduler] [Info]: Initializing mapping policy
[Scheduler] [Info]: Initializing DVFS policy
Executing Python script /home/roozbeh/projects/HotSniper/HotSniper/benchmarks/sim.scripts.py

[Scheduler]: Trying to map Thread  0 from Task 0 at Time 0 ns

[Scheduler]: Trying to schedule Task 0 at Time 0 ns

[Scheduler]: Task 0 put into execution queue. 
[Scheduler]: Assigning Core 0 to Task 0
[Scheduler]: Assigning Core 1 to Task 0
[Scheduler]: Assigning Core 2 to Task 0
[Scheduler]: Assigning Core 3 to Task 0
[Scheduler]: Assigning Core 4 to Task 0
[Scheduler]: Assigning Core 5 to Task 0
[Scheduler]: Assigning Core 6 to Task 0
[Scheduler]: Assigning Core 7 to Task 0
[Scheduler]: Assigning Core 8 to Task 0
[Scheduler]: Assigning Core 9 to Task 0
[Scheduler]: Assigning Core 10 to Task 0
[Scheduler]: Assigning Core 11 to Task 0
[Scheduler]: Assigning Core 12 to Task 0
[Scheduler]: Assigning Core 13 to Task 0
[Scheduler]: Assigning Core 14 to Task 0
[Scheduler]: Assigning Core 15 to Task 0

[Scheduler]: Setting Affinity for Thread 0 from Task 0 to Core 0

[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Sniper using SIFT/trace-driven frontend
[SNIPER] Running full application in DETAILED mode
[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Enabling performance models
[SNIPER] Setting instrumentation mode to DETAILED
[app0] PARSEC Benchmark Suite Version 2.1
[app0] [HOOKS] PARSEC Hooks Version 1.2

[Scheduler]: Time 1.000.000 ns [Active Tasks =  1 | Completed Tasks = 0 | Queued Tasks = 0 | Non-Queued Tasks  = 0 | Free Cores = 48 | Active Tasks Requirements = 16 ] 

[Scheduler]: DVFS Control Loop invoked at 1.000.000 ns
[Scheduler][DVFS_MAX_FREQ]: Core  0: P=-1.000 W f=4000 MHz T=-1.0 °C utilization=1.000
[SNIPER] Setting frequency for core 0 in DVFS domain 0 to 1000 MHz
[SNIPER] Setting frequency for core 1 in DVFS domain 1 to 1000 MHz
[SNIPER] Setting frequency for core 2 in DVFS domain 2 to 1000 MHz
[SNIPER] Setting frequency for core 3 in DVFS domain 3 to 1000 MHz
[SNIPER] Setting frequency for core 4 in DVFS domain 4 to 1000 MHz
[SNIPER] Setting frequency for core 5 in DVFS domain 5 to 1000 MHz
[SNIPER] Setting frequency for core 6 in DVFS domain 6 to 1000 MHz
[SNIPER] Setting frequency for core 7 in DVFS domain 7 to 1000 MHz
[SNIPER] Setting frequency for core 8 in DVFS domain 8 to 1000 MHz
[SNIPER] Setting frequency for core 9 in DVFS domain 9 to 1000 MHz
[SNIPER] Setting frequency for core 10 in DVFS domain 10 to 1000 MHz
[SNIPER] Setting frequency for core 11 in DVFS domain 11 to 1000 MHz
[SNIPER] Setting frequency for core 12 in DVFS domain 12 to 1000 MHz
[SNIPER] Setting frequency for core 13 in DVFS domain 13 to 1000 MHz
[... 124 lines cut ...]
    self.update()
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 90, in update
    power = self.run_power(self.name_last, current)
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 160, in run_power
    execfile(outputbase + '.py', {}, result)
IOError: [Errno 2] No such file or directory: '/home/roozbeh/projects/HotSniper/HotSniper/benchmarks/energystats-temp.py'
83585      [326308]  [ 0]  [py_stats.cc         : 152]  *WARNING* Stats callback: return value must be (convertable into) 64-bit unsigned integer
sh: 1: /home/roozbeh/projects/HotSniper/HotSniper/tools/mcpat.py: Permission denied
Traceback (most recent call last):
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 103, in get_stat
    self.update()
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 90, in update
    power = self.run_power(self.name_last, current)
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 160, in run_power
    execfile(outputbase + '.py', {}, result)
IOError: [Errno 2] No such file or directory: '/home/roozbeh/projects/HotSniper/HotSniper/benchmarks/energystats-temp.py'
98451      [326308]  [ 0]  [py_stats.cc         : 152]  *WARNING* Stats callback: return value must be (convertable into) 64-bit unsigned integer
sh: 1: /home/roozbeh/projects/HotSniper/HotSniper/tools/mcpat.py: Permission denied
Traceback (most recent call last):
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 103, in get_stat
    self.update()
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 90, in update
    power = self.run_power(self.name_last, current)
  File "/home/roozbeh/projects/HotSniper/HotSniper/scripts/energystats.py", line 160, in run_power
    execfile(outputbase + '.py', {}, result)

ma-rapp, 2021-02-01:
Please execute this command in the container and paste the results here:
``$ ls -l tools/mcpat.py``
I would be interested to check the permission flags.

Roozbeh2429, 2021-02-01:
Here is the result:

-rw-rw-r-- 1 1000 1000 104152 Feb  1 12:59 tools/mcpat.py

ma-rapp, 2021-02-01:
It seems the file is owned by root. Did you use ``sudo`` also to clone the repo?

Roozbeh2429, 2021-02-01:
no, only "git clone ..."

ma-rapp, 2021-02-01:
hm... can you try to clone it again, build it, etc.? Importantly, at no step should you be required to use ``sudo``. Maybe the file ownership was compromised by using ``sudo`` at some time.

Roozbeh2429, 2021-02-01:
I built it again with the non-root user docker,(i cloned the repo and built it from scratch) but the same error.

even for running the test on the /test/fft there is a problem with shade libraries

I have no name!@0ce67a0de19e:/home/roozbeh/projects/HotSniper/HotSniper/test/fft$ make run
../../run-sniper -v -n 1 -c gainestown --roi -- ./fft -p 1
[SNIPER] Start
[SNIPER] Running ['bash', '-c', '/home/roozbeh/projects/HotSniper/HotSniper/pin_kit/intel64/bin/pinbin -mt -injection child -ifeellucky -xyzzy -t /home/roozbeh/projects/HotSniper/HotSniper/lib/pin_sim -c /home/roozbeh/projects/HotSniper/HotSniper/config/base.cfg --general/total_cores=1 --general/output_dir=/home/roozbeh/projects/HotSniper/HotSniper/test/fft --config=/home/roozbeh/projects/HotSniper/HotSniper/config/nehalem.cfg --config=/home/roozbeh/projects/HotSniper/HotSniper/config/gainestown.cfg -g --general/magic=true -g --traceinput/stop_with_first_app=true -g --traceinput/restart_apps=false -- ./fft -p 1']
/home/roozbeh/projects/HotSniper/HotSniper/pin_kit/intel64/bin/pinbin: error while loading shared libraries: libstlport-dynamic.so: cannot open shared object file: No such file or directory
[SNIPER] End
[SNIPER] Elapsed time: 0.00 seconds
Makefile:8: recipe for target 'run_fft' failed
make: *** [run_fft] Error 127

anujpathania, 2021-02-01:
Hi. Are you able to run default Sniper (with McPat Enabled) without any issue?

Roozbeh2429, 2021-02-01:
Hi, I did that, but not exactly without issue. I made the changes at this like and it worked:

[https://github.com/farabimahmud/sniper-simulator-7.2-pin-3.11-97998](https://github.com/farabimahmud/sniper-simulator-7.2-pin-3.11-97998)

anujpathania, 2021-02-01:
Sniper default tests like FFT are not well-tested with HotSniper. The focus of this tool is on PARSEC benchmarks.

anujpathania, 2021-02-01:
Coming back to the permission issue, it seems like a machine-specific privilege issue. We don't have anymore suggestions to offer. If you do resolve this issue, please do remember to post the solution in this thread also.

Roozbeh2429, 2021-02-01:
ok sure. thank you for your help.

MohsenAnsari71, 2021-02-01:
> Sniper default tests like FFT are not well-tested with HotSniper. The focus of this tool is on PARSEC benchmarks.

Dear Anuj, I cannot understand why it does not work for other benchmarks like the MiBench benchmark suite?

anujpathania, 2021-02-01:
To be precise, the "open" scheduler we designed doesn't work with benchmarks other than PARSEC and SPLASH because it requires information about how much maximum threads in parallel each benchmark will run. See the file "/master/common/scheduler/scheduler_open.cc" (Line 800 - 874). That information is already  coded in for the PARSEC and SPLASH benchmarks. If you add that info for MiBench benchmark than it will also work.

In current implementation, Open Scheduler only lets in a task when number of free cores are more than its maximum parallel core requirement because it supports only one thread per core model and if number of active threads become more than total cores it can't handle that scenario. Ofcourse, code can always be improved to handle that case but we have not done it yet.

If you use any of other default scheduler like static this limitation will not apply, and any binary should ideally work. But Open scheduler magnitude more sophisticated than default schedulers in other aspects.

MohsenAnsari71, 2021-02-01:
Thanks,
I think you are talking about the mapping strategies, not scheduling policies. Which scheduling policies are supported via this tool? I think you assumed that the maximum number of threads in the system is less than the number of cores, then you do not need any scheduling policies. Am I right?

Roozbeh2429, 2021-02-02:
> Coming back to the permission issue, it seems like a machine-specific privilege issue. We don't have anymore suggestions to offer. If you do resolve this issue, please do remember to post the solution in this thread also.

The permission issue resolved by giving execution permission to `HotSniper/tools/mcpat.py`.

Roozbeh2429, 2021-02-02:
But steel this command raises an error,
This time, in the end, it makes the result folder with some results in it, but I don't know that with these errors, are the results correct or not?
 The error is :  
`../hotspot/hotspot: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by ../hotspot/hotspot)`

Here is the first part of the console log:

```
I have no name!@0cf17b3d893e:/home/roozbeh/projects/HotSniper/HotSniper/simulationcontrol$ PYTHONIOENCODING="UTF-8" python3 run.py
running parsec-blackscholes-simsmall-15 with configuration 1.0GHz+maxFreq+slowDVFS
-n 64 -c gainestown --benchmarks=parsec-blackscholes-simsmall-15 --no-roi --sim-end=last -senergystats:1000000 -speriodic-power:1000000
[app0] [PARSEC] Benchmarks to run: blackscholes
[app0] 
[app0] [PARSEC] [========== Running benchmark blackscholes ==========]
[app0] [PARSEC] Setting up run directory.
[app0] [PARSEC] Unpacking benchmark input 'simsmall'.
[app0] in_4K.txt
[SNIPER] Start
[app0] [PARSEC] Running '/home/roozbeh/projects/HotSniper/HotSniper/record-trace --routine-tracing -o /tmp/tmp9Cm8oM/run_benchmarks -e 1 -r 1 -s 0 -- /home/roozbeh/projects/HotSniper/HotSniper/benchmarks/parsec/parsec-2.1/pkgs/apps/blackscholes/inst/amd64-linux.gcc-sniper/bin/blackscholes 15 in_4K.txt prices.txt':
[app0] [PARSEC] [---------- Beginning of output ----------]
[Scheduler]: Setting Arrival Time for Task 0 (parsec-blackscholes-simsmall-15) to 0 ns
[Scheduler] [Info]: Initializing mapping policy
[Scheduler] [Info]: Initializing DVFS policy
Executing Python script /home/roozbeh/projects/HotSniper/HotSniper/benchmarks/sim.scripts.py

[Scheduler]: Trying to map Thread  0 from Task 0 at Time 0 ns

[Scheduler]: Trying to schedule Task 0 at Time 0 ns

[Scheduler]: Task 0 put into execution queue. 
[Scheduler]: Assigning Core 0 to Task 0
[Scheduler]: Assigning Core 1 to Task 0
[Scheduler]: Assigning Core 2 to Task 0
[Scheduler]: Assigning Core 3 to Task 0
[Scheduler]: Assigning Core 4 to Task 0
[Scheduler]: Assigning Core 5 to Task 0
[Scheduler]: Assigning Core 6 to Task 0
[Scheduler]: Assigning Core 7 to Task 0
[Scheduler]: Assigning Core 8 to Task 0
[Scheduler]: Assigning Core 9 to Task 0
[Scheduler]: Assigning Core 10 to Task 0
[Scheduler]: Assigning Core 11 to Task 0
[Scheduler]: Assigning Core 12 to Task 0
[Scheduler]: Assigning Core 13 to Task 0
[Scheduler]: Assigning Core 14 to Task 0
[Scheduler]: Assigning Core 15 to Task 0

[Scheduler]: Setting Affinity for Thread 0 from Task 0 to Core 0

[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Sniper using SIFT/trace-driven frontend
[SNIPER] Running full application in DETAILED mode
[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Enabling performance models
[SNIPER] Setting instrumentation mode to DETAILED
[app0] PARSEC Benchmark Suite Version 2.1
[app0] [HOOKS] PARSEC Hooks Version 1.2

[Scheduler]: Time 1.000.000 ns [Active Tasks =  1 | Completed Tasks = 0 | Queued Tasks = 0 | Non-Queued Tasks  = 0 | Free Cores = 48 | Active Tasks Requirements = 16 ] 

[Scheduler]: DVFS Control Loop invoked at 1.000.000 ns
[Scheduler][DVFS_MAX_FREQ]: Core  0: P=5.281 W f=4000 MHz T=-1.0 °C utilization=0.472
[SNIPER] Setting frequency for core 0 in DVFS domain 0 to 1000 MHz
[SNIPER] Setting frequency for core 1 in DVFS domain 1 to 1000 MHz
[SNIPER] Setting frequency for core 2 in DVFS domain 2 to 1000 MHz
[SNIPER] Setting frequency for core 3 in DVFS domain 3 to 1000 MHz
[SNIPER] Setting frequency for core 4 in DVFS domain 4 to 1000 MHz
[SNIPER] Setting frequency for core 5 in DVFS domain 5 to 1000 MHz
[SNIPER] Setting frequency for core 6 in DVFS domain 6 to 1000 MHz
[SNIPER] Setting frequency for core 7 in DVFS domain 7 to 1000 MHz
[... 91 lines cut ...]
    main(jobid = jobid, resultsdir = resultsdir, powertype = powertype, config = config, outputfile = outputfile, no_graph = no_graph, print_stack = not no_text, partial = partial)
  File "/home/roozbeh/projects/HotSniper/HotSniper/tools/mcpat.py", line 310, in main
    results = power_stack(power_dat,results['config'], powertype)
  File "/home/roozbeh/projects/HotSniper/HotSniper/tools/mcpat.py", line 616, in power_stack
    temperatures = subprocess.check_output([hotspot_binary] + hotspot_args)
  File "/usr/lib/python2.7/subprocess.py", line 574, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['../hotspot/hotspot', '-c', '../hotspot/hotspot.config', '-f', '/home/roozbeh/projects/HotSniper/HotSniper/benchmarks/8x8_manycore.flp', '-sampling_intvl', '0.001', '-p', 'InstantaneousPower.log', '-o', 'InstantaneousTemperature.log', '-init_file', 'Temperature.init']' returned non-zero exit status 1

[Scheduler]: Time 4.000.000 ns [Active Tasks =  1 | Completed Tasks = 0 | Queued Tasks = 0 | Non-Queued Tasks  = 0 | Free Cores = 48 | Active Tasks Requirements = 16 ] 

[Scheduler]: DVFS Control Loop invoked at 4.000.000 ns
[Scheduler][DVFS_MAX_FREQ]: Core  0: P=0.555 W f=1000 MHz T=-1.0 °C utilization=0.385
../hotspot/hotspot: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by ../hotspot/hotspot)
Traceback (most recent call last):
  File "/home/roozbeh/projects/HotSniper/HotSniper/tools/mcpat.py", line 1601, in <module>
[... cut ...]
    temperatures = subprocess.check_output([hotspot_binary] + hotspot_args)
  File "/usr/lib/python2.7/subprocess.py", line 574, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['../hotspot/hotspot', '-c', '../hotspot/hotspot.config', '-f', '/home/roozbeh/projects/HotSniper/HotSniper/benchmarks/8x8_manycore.flp', '-sampling_intvl', '0.001', '-p', 'InstantaneousPower.log', '-o', 'InstantaneousTemperature.log', '-init_file', 'Temperature.init']' returned non-zero exit status 1
```

anujpathania, 2021-02-02:
No, execution is incomplete. I believe the C libraries require to execute (compile) Hotspot are missing on your machine. Maybe you can download and original Hotspot and see if you can compile and execute that successfully.

ma-rapp, 2021-02-02:
> The permission issue resolved by giving execution permission to `HotSniper/tools/mcpat.py`.

You are right. Thanks for spotting this. The executable flag was accidentally reset during the recent changes. I fixed it.

This issue has now several questions mixed in one. To avoid confusion, I suggest to keep this issue for the "Permission denied" error. Please close the issue if this error is fixed.

If you encounter another problem, I suggest to open a separate issue.

closed 2021-02-02 by anujpathania