anujpathania/HotSniper #3: step 2 "Compiling HotSniper7" make

Roozbeh2429, 2021-02-01:
Hi. I tried to build Hot-sniper on three different Linux machines with three fresh, **newly installed** Ubuntu. versions (16.04, 18.04, 20.04) . the `$ make`  command for making the docker container raises the same error in all of them, at the same part: 
I could make this docker container successfully about two months ago, but now it has this issue.
Maybe recently, something changed on dependencies.

```
~/projects/HotSniper/docker$ make
docker build -f Dockerfile-ubuntu-16.04 -t ubuntu:16.04-sniper-roozbeh .
Sending build context to Docker daemon  6.656kB
Step 1/10 : FROM ubuntu:16.04
 ---> 8185511cd5ad
Step 2/10 : RUN dpkg --add-architecture i386
 ---> Using cache
 ---> 289743de0922
Step 3/10 : RUN apt-get update && apt-get install -y     python     screen     tmux     binutils     libc6:i386     libncurses5:i386     libstdc++6:i386  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> f4b58aaa0c37
Step 4/10 : RUN apt-get update && apt-get install -y     automake     build-essential     curl     wget     libboost-dev     libsqlite3-dev     zlib1g-dev     libbz2-dev     g++-4.8     g++-4.9  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 2460b94b5113
Step 5/10 : RUN apt-get update && apt-get install -y     nano     autoconf     automake     autotools-dev     bc     bison     curl     device-tree-compiler     flex     gawk     gperf     libexpat-dev     libgmp-dev     libmpc-dev     libmpfr-dev     libtool     libusb-1.0-0-dev     patchutils     pkg-config     texinfo     zlib1g-dev  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 94990987004b
Step 6/10 : RUN apt-get update && apt-get install -y     gdb     git  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 29d3e1e81d09
Step 7/10 : RUN apt-get update && apt-get install -y     gettext     gfortran     m4     xsltproc     libx11-dev     libxext-dev     libxt-dev     libxmu-dev     libxi-dev     gnuplot  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> e5df49cf153a
Step 8/10 : RUN apt-get update && apt-get install -y     python3     python3-setuptools  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 224fe2c4be3e
Step 9/10 : RUN easy_install3 pip  && rm -rf /var/lib/apt/lists/*
 ---> Running in f24a58f4e2e0
Searching for pip
Reading https://pypi.python.org/simple/pip/
Best match: pip 21.0.1
Downloading https://files.pythonhosted.org/packages/b7/2d/ad02de84a4c9fd3b1958dc9fb72764de1aa2605a9d7e943837be6ad82337/pip-21.0.1.tar.gz#sha256=99bbde183ec5ec037318e774b0d8ae0a64352fe53b2c7fd630be1d07e94f41e5
Processing pip-21.0.1.tar.gz
Writing /tmp/easy_install-f79qp5a5/pip-21.0.1/setup.cfg
Running pip-21.0.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-f79qp5a5/pip-21.0.1/egg-dist-tmp-6velqxmb
/usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'python_requires'
  warnings.warn(msg)
/usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'project_urls'
  warnings.warn(msg)
warning: no files found matching 'docs/docutils.conf'
warning: no previously-included files found matching '.coveragerc'
warning: no previously-included files found matching '.mailmap'
warning: no previously-included files found matching '.appveyor.yml'
warning: no previously-included files found matching '.travis.yml'
warning: no previously-included files found matching '.readthedocs.yml'
warning: no previously-included files found matching '.pre-commit-config.yaml'
warning: no previously-included files found matching 'tox.ini'
warning: no previously-included files found matching 'noxfile.py'
warning: no files found matching '*.css' under directory 'docs'
warning: no previously-included files found matching 'src/pip/_vendor/six'
warning: no previously-included files found matching 'src/pip/_vendor/six/moves'
warning: no previously-included files matching '*.pyi' found under directory 'src/pip/_vendor'
no previously-included directories found matching '.github'
no previously-included directories found matching '.azure-pipelines'
no previously-included directories found matching 'docs/build'
no previously-included directories found matching 'news'
no previously-included directories found matching 'tasks'
no previously-included directories found matching 'tests'
no previously-included directories found matching 'tools'
  File "build/bdist.linux-x86_64/egg/pip/_internal/vcs/versioncontrol.py", line 72
    req = f'{repo_url}@{rev}#egg={egg_project_name}'
                                                   ^
SyntaxError: invalid syntax

  File "build/bdist.linux-x86_64/egg/pip/_internal/vcs/git.py", line 156
[... 482 lines cut ...]
SyntaxError: invalid syntax

Removing intermediate container f24a58f4e2e0
 ---> a48478a5579f
Step 10/10 : RUN pip3 install matplotlib seaborn diskcache tabulate  && rm -rf /var/lib/apt/lists/*
 ---> Running in e9a833d3d6df
Traceback (most recent call last):
  File "/usr/local/bin/pip3", line 9, in <module>
    load_entry_point('pip==21.0.1', 'console_scripts', 'pip3')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 542, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2569, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2229, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.5/dist-packages/pip-21.0.1-py3.5.egg/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
The command '/bin/sh -c pip3 install matplotlib seaborn diskcache tabulate  && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 1
Makefile:6: recipe for target 'all' failed
make: *** [all] Error 1
```

ma-rapp, 2021-02-02:
pip 21 was released one week ago that dropped support for python 3.5 (which is used in Ubuntu 16.04).

I updated the Dockerfile to use pip 20. Please try it again.

Roozbeh2429, 2021-02-02:
Dear Martin, the problem resolved. Thank you for your assist.

closed 2021-02-02 by Roozbeh2429