anujpathania/HotSniper #4: version `GLIBC_2.29' not found

Roozbeh2429, 2021-02-02:
Hi, HotSniper is working well on Ubuntu 16.04. but on **Ubuntu 20.04.2 LTS** it seems that the **hotspot** has some problems.
It will be great if the HotSniper becomes functional on **Ubuntu 20.04** too. 

The error is:
`../hotspot/hotspot: /lib/x86_64-linux-gnu/libm.so.6: version 'GLIBC_2.29' not found (required by ../hotspot/hotspot)`

Here is the first part of the console log:

```
I have no name!@60cd8fd785f0:/home/roozbeh/projects/HotSniper/simulationcontrol$ PYTHONIOENCODING="UTF-8" python3 run.py
running parsec-blackscholes-simsmall-15 with configuration 1.0GHz+maxFreq+slowDVFS
-n 64 -c gainestown --benchmarks=parsec-blackscholes-simsmall-15 --no-roi --sim-end=last -senergystats:1000000 -speriodic-power:1000000
[app0] [PARSEC] Benchmarks to run: blackscholes
[app0] 
[app0] [PARSEC] [========== Running benchmark blackscholes ==========]
[app0] [PARSEC] Setting up run directory.
[app0] [PARSEC] Unpacking benchmark input 'simsmall'.
[SNIPER] Start
[app0] in_4K.txt
[app0] [PARSEC] Running '/home/roozbeh/projects/HotSniper/record-trace --routine-tracing -o /tmp/tmpOOhgsO/run_benchmarks -e 1 -r 1 -s 0 -- /home/roozbeh/projects/HotSniper/benchmarks/parsec/parsec-2.1/pkgs/apps/blackscholes/inst/amd64-linux.gcc-sniper/bin/blackscholes 15 in_4K.txt prices.txt':
[app0] [PARSEC] [---------- Beginning of output ----------]
[Scheduler]: Setting Arrival Time for Task 0 (parsec-blackscholes-simsmall-15) to 0 ns
[Scheduler] [Info]: Initializing mapping policy
[Scheduler] [Info]: Initializing DVFS policy
Executing Python script /home/roozbeh/projects/HotSniper/benchmarks/sim.scripts.py

[Scheduler]: Trying to map Thread  0 from Task 0 at Time 0 ns

[Scheduler]: Trying to schedule Task 0 at Time 0 ns

[Scheduler]: Task 0 put into execution queue. 
[Scheduler]: Assigning Core 0 to Task 0
[Scheduler]: Assigning Core 1 to Task 0
[Scheduler]: Assigning Core 2 to Task 0
[Scheduler]: Assigning Core 3 to Task 0
[Scheduler]: Assigning Core 4 to Task 0
[Scheduler]: Assigning Core 5 to Task 0
[Scheduler]: Assigning Core 6 to Task 0
[Scheduler]: Assigning Core 7 to Task 0
[Scheduler]: Assigning Core 8 to Task 0
[Scheduler]: Assigning Core 9 to Task 0
[Scheduler]: Assigning Core 10 to Task 0
[Scheduler]: Assigning Core 11 to Task 0
[Scheduler]: Assigning Core 12 to Task 0
[Scheduler]: Assigning Core 13 to Task 0
[Scheduler]: Assigning Core 14 to Task 0
[Scheduler]: Assigning Core 15 to Task 0

[Scheduler]: Setting Affinity for Thread 0 from Task 0 to Core 0

[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Sniper using SIFT/trace-driven frontend
[SNIPER] Running full application in DETAILED mode
[SNIPER] --------------------------------------------------------------------------------
[SNIPER] Enabling performance models
[SNIPER] Setting instrumentation mode to DETAILED
[app0] PARSEC Benchmark Suite Version 2.1
[app0] [HOOKS] PARSEC Hooks Version 1.2

[Scheduler]: Time 1.000.000 ns [Active Tasks =  1 | Completed Tasks = 0 | Queued Tasks = 0 | Non-Queued Tasks  = 0 | Free Cores = 48 | Active Tasks Requirements = 16 ] 

[Scheduler]: DVFS Control Loop invoked at 1.000.000 ns
[Scheduler][DVFS_MAX_FREQ]: Core  0: P=0.547 W f=4000 MHz T=-1.0 °C utilization=0.372
[SNIPER] Setting frequency for core 0 in DVFS domain 0 to 1000 MHz
[SNIPER] Setting frequency for core 1 in DVFS domain 1 to 1000 MHz
[SNIPER] Setting frequency for core 2 in DVFS domain 2 to 1000 MHz
[SNIPER] Setting frequency for core 3 in DVFS domain 3 to 1000 MHz
[SNIPER] Setting frequency for core 4 in DVFS domain 4 to 1000 MHz
[SNIPER] Setting frequency for core 5 in DVFS domain 5 to 1000 MHz
[SNIPER] Setting frequency for core 6 in DVFS domain 6 to 1000 MHz
[... 50 lines cut ...]
[SNIPER] Setting frequency for core 57 in DVFS domain 57 to 1000 MHz
[SNIPER] Setting frequency for core 58 in DVFS domain 58 to 1000 MHz
[SNIPER] Setting frequency for core 59 in DVFS domain 59 to 1000 MHz
[SNIPER] Setting frequency for core 60 in DVFS domain 60 to 1000 MHz
[SNIPER] Setting frequency for core 61 in DVFS domain 61 to 1000 MHz
[SNIPER] Setting frequency for core 62 in DVFS domain 62 to 1000 MHz
[SNIPER] Setting frequency for core 63 in DVFS domain 63 to 1000 MHz
../hotspot/hotspot: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by ../hotspot/hotspot)
Traceback (most recent call last):
  File "/home/roozbeh/projects/HotSniper/tools/mcpat.py", line 1601, in <module>
    main(jobid = jobid, resultsdir = resultsdir, powertype = powertype, config = config, outputfile = outputfile, no_graph = no_graph, print_stack = not no_text, partial = partial)
  File "/home/roozbeh/projects/HotSniper/tools/mcpat.py", line 310, in main
    results = power_stack(power_dat,results['config'], powertype)
  File "/home/roozbeh/projects/HotSniper/tools/mcpat.py", line 616, in power_stack
    temperatures = subprocess.check_output([hotspot_binary] + hotspot_args)
  File "/usr/lib/python2.7/subprocess.py", line 574, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['../hotspot/hotspot', '-c', '../hotspot/hotspot.config', '-f', '/home/roozbeh/projects/HotSniper/benchmarks/8x8_manycore.flp', '-sampling_intvl', '0.001', '-p', 'InstantaneousPower.log', '-o', 'InstantaneousTemperature.log']' returned non-zero exit status 1

[Scheduler]: Time 2.000.000 ns [Active Tasks =  1 | Completed Tasks = 0 | Queued Tasks = 0 | Non-Queued Tasks  = 0 | Free Cores = 48 | Active Tasks Requirements = 16 ] 

[Scheduler]: DVFS Control Loop invoked at 2.000.000 ns
[Scheduler][DVFS_MAX_FREQ]: Core  0: P=0.545 W f=1000 MHz T=-1.0 °C utilization=0.370
../hotspot/hotspot: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by ../hotspot/hotspot)
```

ma-rapp, 2021-02-02:
I guess you built HotSpot outside docker like described in the README. Could you try building hotspot inside docker?

Roozbeh2429, 2021-02-02:
Dear Martin, the problem resolved by **building the hotspot inside the docker container**. thank you so much.

closed 2021-02-02 by Roozbeh2429

ma-rapp, 2021-02-03:
I updated the README