anujpathania/HotSniper #5: HotSniper/scripts

Roozbeh2429, 2021-02-04:
Hi. What is the best way for using scripts like `powertrace.py` or `dvfs.py`, ... that exists on `HotSniper/scripts` with PARSEC benchmarks?
On the default Sniper, it is possible to run the below command on the `sniper/benchmarks` :

`./run-sniper -p parsec-dedup -i small -n 4 -c gainestown -d ../res -s powertrace.py`

but on `HotSniper/benchmarks`, this command raises an error:

```
I have no name!@cacdc1531d65:/home/roozbeh/projects/HotSniper/benchmarks$ ./run-sniper -p parsec-dedup -i small -n 4 -c gainestown -d ../res -s powertrace.py
[PARSEC] Benchmarks to run:  dedup

[PARSEC] [========== Running benchmark dedup ==========]
[PARSEC] Setting up run directory.
[PARSEC] Unpacking benchmark input 'simsmall'.
media.dat
[PARSEC] Running 'SNIPER_APP_LD_PRELOAD=; unset LD_PRELOAD; /home/roozbeh/projects/HotSniper/run-sniper -n 4 -m 'localhost' -d '/home/roozbeh/projects/HotSniper/res' -c gainestown -s powertrace.py --roi --curdir=/home/roozbeh/projects/HotSniper/benchmarks  --  /home/roozbeh/projects/HotSniper/benchmarks/parsec/parsec-2.1/pkgs/kernels/dedup/inst/amd64-linux.gcc-sniper/bin/dedup -c -p -f -t 1 -i media.dat -o output.dat.ddp':
[PARSEC] [---------- Beginning of output ----------]
[SNIPER] Start
/home/roozbeh/projects/HotSniper/pin_kit/intel64/bin/pinbin: error while loading shared libraries: libstlport-dynamic.so: cannot open shared object file: No such file or directory
[SNIPER] End
[SNIPER] Elapsed time: 0.00 seconds
[PARSEC] [----------    End of output    ----------]
[PARSEC] Done.
```

ma-rapp, 2021-02-04:
I have seen this error before. It has nothing to do with the scripts but with how the benchmark is specified. Please try to pass benchmarks with the ``--benchmarks`` option:

in your case:
<pre>
./run-sniper --benchmarks=parsec-dedup-simsmall-4 -c gainestown -d ../res -s powertrace.py
</pre>

Roozbeh2429, 2021-02-04:
Dear Martin, the error changed. but steel there is an error with this command:

```
I have no name!@d257cf1d929d:/home/roozbeh/projects/HotSniper/benchmarks$ ./run-sniper --benchmarks=parsec-dedup-simsmall-4 -c gainestown -d ../../res -s powertrace.py
[app0] [PARSEC] Benchmarks to run: dedup
[app0] 
[app0] [PARSEC] [========== Running benchmark dedup ==========]
[app0] [PARSEC] Setting up run directory.
[app0] [PARSEC] Unpacking benchmark input 'simsmall'.
[app0] media.dat
[SNIPER] Start
[app0] [PARSEC] Running '/home/roozbeh/projects/HotSniper/record-trace --routine-tracing -o /tmp/tmp28VR5j/run_benchmarks -e 1 -r 1 -s 0 --roi -- /home/roozbeh/projects/HotSniper/benchmarks/parsec/parsec-2.1/pkgs/kernels/dedup/inst/amd64-linux.gcc-sniper/bin/dedup -c -p -f -t 4 -i media.dat -o output.dat.ddp':
[app0] [PARSEC] [---------- Beginning of output ----------]
Assertion error in thermal model file: file ended too early.
[SNIPER] End
[SNIPER] Elapsed time: 0.18 seconds
```

ma-rapp, 2021-02-04:
There is an error in your thermal model configuration. The configuration checklist in the README should help you resolve it.

Roozbeh2429, 2021-02-05:
Thank you a lot.

closed 2021-02-05 by Roozbeh2429