anujpathania/HotSniper #6: Error in running a benchmark

anishrajghimire, 2021-02-18:
Hello,
I compiled the HotSniper which works fine. But when I run a benchmark, it gives an error given. It would be great if anyone can help.

anish@debian10:~/HotSniper-master/benchmarks$ ./run-sniper --benchmarks=parsec-dedup-simsmall-4 -c gainestown -d ../res -s powertrace.py
[app0] [PARSEC] Benchmarks to run: dedup
[app0] 
[app0] [PARSEC] [========== Running benchmark dedup ==========]
[SNIPER] Warning: Unable to use physical addresses for shared memory simulation.
[app0] [PARSEC] Setting up run directory.
[app0] [PARSEC] Unpacking benchmark input 'simsmall'.
[app0] media.dat
[app0] [PARSEC] Running '/home/anish/HotSniper-master/record-trace --routine-tracing -o /tmp/tmpLnV3rM/run_benchmarks -e 1 -r 1 -s 0 --roi -- /home/anish/HotSniper-master/benchmarks/parsec/parsec-2.1/pkgs/kernels/dedup/inst/amd64-linux.gcc-sniper/bin/dedup -c -p -f -t 4 -i media.dat -o output.dat.ddp':
[app0] [PARSEC] [---------- Beginning of output ----------]
[SNIPER] Start
[app0] E: Unable to load /home/anish/HotSniper-master/sift/recorder/obj-intel64/sift_recorder: dlopen failed: cannot locate symbol "_ZnwmSt11align_val_t" referenced by "/home/anish/HotSniper-master/sift/recorder/obj-intel64/sift_recorder"...
[app0] [RECORD-TRACE] Using the Pin frontend (sift/recorder)
[app0] [PARSEC] [---------- End of output ----------]
[app0] [PARSEC] Done.
Assertion error in thermal model file: numberUnits != coreRows * coreColumns
[SNIPER] End
[SNIPER] Elapsed time: 0.17 seconds
[RECORD-TRACE] Using the Pin frontend (sift/recorder)
E: Missing application name
Pin: pin-3.7-97619-0d0c92f4f
Copyright (c) 2003-2018, Intel Corporation. All rights reserved.

Usage: pin [OPTION] [-t <tool> [<toolargs>]] -- <command line>
Use -help for a description of options
anish@debian10:~/HotSniper-master/benchmarks$

anujpathania, 2021-02-21:
You need to also mention the number of cores in the command line for this command.

For example, try "./run-sniper --benchmarks=parsec-dedup-simsmall-4 -c gainestown -n 64 -d ../res -s powertrace.py"

anujpathania, 2021-02-21:
It is better to run benchmarks using the Simulation control instead, as mentioned in the ReadMe. Also take a look at the Configuration Checklist.

Also don't forget to close the issue if and when your problem gets resolved. Thanks.

anujpathania, 2021-02-24:
I am closing this issue. Please reopen this issue, if your problem persists.

closed 2021-02-24 by anujpathania