anujpathania/HotSniper #7: help wanted: Could not configure 'libc6:i386'

imsridhar, 2021-02-20:
After "sudo make" in the second step (make in docker folder)
I am getting this-
///////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Processing triggers for libc-bin (2.31-0ubuntu9.1) ...
E: Could not configure 'libc6:i386'. 
E: Could not perform immediate configuration on 'libgcc-s1:i386'. Please see man 5 apt.conf under APT::Immediate-Configure for details. (2)
The command '/bin/sh -c apt-get update && apt-get install -y     python     screen     tmux     binutils     libc6:i386     libncurses5:i386     libstdc++6:i386  && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
make: *** [Makefile:6: all] Error 100
///////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Below are the solution I tried(from various sources)
///////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Try(1): 'sudo apt-get install --reinstall libc6-i386'
remark: not working after this
------------------------------------------------------------------------------------------
Try(2)
-> sudo docker pull ubuntu:20.04
-> sudo docker run --rm ubuntu:20.04 sh -c 'dpkg --add-architecture i386 && apt-get update -qq && apt-get install -y libc6:i386'

source -> https://bugs.launchpad.net/ubuntu-cdimage/+bug/1871268
source-> https://bugs.launchpad.net/ubuntu-cdimage/+bug/1871268/comments/39

output :
Processing triggers for libc-bin (2.31-0ubuntu9.1) ...
E: Could not configure 'libc6:i386'. 
E: Could not perform immediate configuration on 'libgcc-s1:i386'. Please see man 5 apt.conf under APT::Immediate-Configure for details. (2)

remark: not working
------------------------------------------------------------------------------------------
Note: Minimal python is already installed (re-installed too)
------------------------------------------------------------------------------------------

anujpathania, 2021-02-21:
It seems like a docker issue. 

Make commands should not be executed with sudo. Docker should also be run without sudo.

Are you able to, without sudo, run  command "docker run hello-world"?

imsridhar, 2021-02-21:
No, it doesn't work without sudo.

anujpathania, 2021-02-21:
I just tested it on a fresh installation of Ubuntu 20.04 LTS and it (including docker) works fine but not if you use sudo. 

You may have to add your user to the group to work without Sudo.

For example, try command “sudo usermod -aG docker $USER” and restart Ubuntu.

In any case, you must be able to run Docker Hello-World without sudo first in your environment. Then only we can assist further.

anujpathania, 2021-02-24:
I am closing this issue. for now. Please reopen this issue, if your problem persists.

closed 2021-02-24 by anujpathania