anujpathania/HotSniper #10: Empty HotSniper folder when running in docker image

gkothar1, 2021-04-01:
After following these steps:
```
cd docker
make
make run
```
Docker is successfully set up as per the log:
```
docker build -f Dockerfile-ubuntu-16.04 -t ubuntu:16.04-sniper-root .
Sending build context to Docker daemon  6.656kB
Step 1/10 : FROM ubuntu:16.04
 ---> f6f49faac5cf
Step 2/10 : RUN dpkg --add-architecture i386
 ---> Using cache
 ---> baf73252e6f5
Step 3/10 : RUN apt-get update && apt-get install -y     python     screen     tmux     binutils     libc6:i386     libncurses5:i386     libstdc++6:i386  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 5d9a76dbb756
Step 4/10 : RUN apt-get update && apt-get install -y     automake     build-essential     curl     wget     libboost-dev     libsqlite3-dev     zlib1g-dev     libbz2-dev     g++-4.8     g++-4.9  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> d6233aeb8519
Step 5/10 : RUN apt-get update && apt-get install -y     nano     autoconf     automake     autotools-dev     bc     bison     curl     device-tree-compiler     flex     gawk     gperf     libexpat-dev     libgmp-dev     libmpc-dev     libmpfr-dev     libtool     libusb-1.0-0-dev     patchutils     pkg-config     texinfo     zlib1g-dev  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 58f392e0101c
Step 6/10 : RUN apt-get update && apt-get install -y     gdb     git  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 22fea0ea133f
Step 7/10 : RUN apt-get update && apt-get install -y     gettext     gfortran     m4     xsltproc     libx11-dev     libxext-dev     libxt-dev     libxmu-dev     libxi-dev     gnuplot  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 01fb031a03fe
Step 8/10 : RUN apt-get update && apt-get install -y     python3     python3-pip  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> cfb32299ef07
Step 9/10 : RUN pip3 install --upgrade pip==20.*  # pip 21 is not compatible with Ubuntu 16.04 Python version
 ---> Using cache
 ---> 4da002eabbbf
Step 10/10 : RUN pip3 install matplotlib seaborn diskcache tabulate
 ---> Using cache
 ---> 054a50e19727
Successfully built 054a50e19727
Successfully tagged ubuntu:16.04-sniper-roo
```

But when I do `cd ..` to complile rest of the stuff, I see empty HotSniper directory. Anyone came across this issue?

SobhanNiknam, 2021-04-01:
Dear Gaurav,

To solve this issue, you can manually modify the path to the HotSniper directory in your system in the Makefile of Docker. To do so, you should replace "${HOME}:${HOME}" with "path/to/HotSniper:path/to/HotSniper".

Best Regards,
Sobhan

gkothar1, 2021-04-01:
It worked. Thanks, Sobhan!

gkothar1, 2021-04-01:
Can we run the tool without using docker?

anujpathania, 2021-04-02:
Dear Gaurav,

Unfortunately, we do not support execution of HotSniper without Docker anymore. Thanks to Docker, we no longer have to worry about HotSniper's dependencies being broken with an Ubuntu update. Docker also provides compatibility across Ubuntu versions.

I am closing this issue. If you face more technical problems, please feel free to open a new issue. Thanks.

-- Anuj

closed 2021-04-02 by anujpathania

gkothar1, 2021-04-02:
Makes sense, Thanks Anuj!

Regards,
Gaurav

anujpathania, 2022-06-01:
This empty docker problem occurs when you do "make" and "make run" in docker folder using "sudo".

After installing docker from here - https://docs.docker.com/engine/install/ubuntu/

Follow instructions here - https://docs.docker.com/engine/install/linux-postinstall/

This will allow you to run docker without sudo.