anujpathania/HotSniper #11: Not able to use splash2 benchmarks

imsridhar, 2021-05-21:
`running splash2-barnes-simsmall-16 with configuration 1.0GHz+maxFreq+slowDVFS
-n 64 -c gainestown --benchmarks=splash2-barnes-simsmall-16 --no-roi --sim-end=last -senergystats:1000000 -speriodic-power:1000000`

`Invalid input size simsmall`

`Traceback (most recent call last):
  File "/home/sridhar/Documents/csg/Hotsniper/HotSniper/benchmarks/run-sniper", line 291, in <module>
    program = __import__(package).Program(programname, int(app_nthreads), inputsize, ['force_nthreads'])
  File "/home/sridhar/Documents/csg/Hotsniper/HotSniper/benchmarks/splash2/__init__.py", line 28, in __init__
    raise ValueError("Invalid input size %s" % inputsize)`

`ValueError: Invalid input size simsmall
creating plots for results_2021-05-21_11.15_1.0GHz+maxFreq+slowDVFS_splash2-barnes-simsmall-16
Traceback (most recent call last):
  File "run.py", line 248, in <module>
    main()
  File "run.py", line 243, in main
         example()
  File "run.py", line 235, in exslmple owDVFS'], get_instance(benchmark, parallelism, input_set='simsmall'))
  File "run.py", line 114, in run
    raise Exception('return code != 0')`

`Exception: return code != 0`

anujpathania, 2021-05-21:
Hi, name for the input in Splash benchmarks is “small”, and not “simsmall”.

Try this modified command - “-n 64 -c gainestown --benchmarks=splash2-barnes-small-16 --no-roi --sim-end=last -senergystats:1000000 -speriodic-power:1000000”

imsridhar, 2021-05-21:
Thank you. 
It is working now. I have edited "_simsmall -> small_"  in _run_ function call. 
Just for clarification, Is it that _simsmall_ works for parsec benchmarks and _small_ for splash2?

anujpathania, 2021-05-21:
Yes.

closed 2021-05-21 by anujpathania