anujpathania/HotSniper #12: Getting an undefined reference error when compiling the benchmarks

AryaFaramarzi, 2021-05-21:
Here is the error I get after running the commands listed in step 3.

I have no name!@ebbb589aad71:/Users/Aria/Desktop/HotSniper/benchmarks$ make
/Users/Aria/Desktop/HotSniper/benchmarks/tools/scripts/checkdependencies.py
make -C tools/hooks
make[1]: Entering directory '/Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks'
./makebuildscripts.py
make[1]: Leaving directory '/Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks'
make -C splash2
make[1]: Entering directory '/Users/Aria/Desktop/HotSniper/benchmarks/splash2'
make -C ../tools/hooks/
make[2]: Entering directory '/Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks'
./makebuildscripts.py
make[2]: Leaving directory '/Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks'
cd splash2/codes; make
make[2]: Entering directory '/Users/Aria/Desktop/HotSniper/benchmarks/splash2/splash2/codes'
make -C /Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks
make[3]: Entering directory '/Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks'
./makebuildscripts.py
make[3]: Leaving directory '/Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks'
for DIR in `find apps kernels -name \[Mm\]akefile|sed -e 's+/[Mm]akefile++'|sort -r`; do CC="gcc" AR="ar" LD="g++" CFLAGS="-g -O3 -I/Users/Aria/Desktop/HotSniper/benchmarks/splash2/splash2/codes -I/Users/Aria/Desktop/HotSniper/benchmarks/splash2/splash2/codes/pthreads -I/Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks -I/Users/Aria/Desktop/HotSniper/include" LDFLAGS="-lm -pthread -uparmacs_roi_end -uparmacs_roi_start -L/Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks -lhooks_base -lrt -pthread" MACROS="/Users/Aria/Desktop/HotSniper/benchmarks/splash2/splash2/codes/pthreads/c.m4.null.POSIX" TARGET_ARCH="" make -C $DIR all; if [ $? -eq 0 ]; then echo; else exit -1; fi; done
make[3]: Entering directory '/Users/Aria/Desktop/HotSniper/benchmarks/splash2/splash2/codes/kernels/radix'
gcc radix.o -g -O3 -I/Users/Aria/Desktop/HotSniper/benchmarks/splash2/splash2/codes -I/Users/Aria/Desktop/HotSniper/benchmarks/splash2/splash2/codes/pthreads -I/Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks -I/Users/Aria/Desktop/HotSniper/include -o RADIX -lm -pthread -uparmacs_roi_end -uparmacs_roi_start -L/Users/Aria/Desktop/HotSniper/benchmarks/tools/hooks -lhooks_base -lrt -pthread
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crt1.o: In function `_start':
(.text+0x20): undefined reference to `main'
collect2: error: ld returned 1 exit status
makefile:7: recipe for target 'RADIX' failed
make[3]: *** [RADIX] Error 1
make[3]: Leaving directory '/Users/Aria/Desktop/HotSniper/benchmarks/splash2/splash2/codes/kernels/radix'
/bin/sh: 1: exit: Illegal number: -1
Makefile:13: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/Users/Aria/Desktop/HotSniper/benchmarks/splash2/splash2/codes'
Makefile:6: recipe for target 'splash2-build' failed
make[1]: *** [splash2-build] Error 2
make[1]: Leaving directory '/Users/Aria/Desktop/HotSniper/benchmarks/splash2'
Makefile:12: recipe for target 'all' failed
make: *** [all] Error 2
I have no name!@ebbb589aad71:/Users/Aria/Desktop/HotSniper/benchmarks$

anujpathania, 2021-05-22:
Hi, Splash benchmarks are failing to compile in your environment.

Are you compiling the HotSniper through docker?

What version of Linux are you using? 

Are you able to compile Splash benchmarks independently outside of HotSniper in your environment?

Does other benchmarks like PARSEC compiling fine?

AryaFaramarzi, 2021-05-23:
Hello, Thanks for the response,
I am compiling through the docker.
ubuntu version 16.04
I can't compile the splash benchmark independently. I get the same error. But other benchmarks ( PARSEC ) will compile just fine.

anujpathania, 2021-05-24:
Hi,

There seem to be an issue with your linker. Make sure bcc version you are using is compatible with SPEC benchmarks.

Resolving SPEC compilation issues not related to HotSniper integration are beyond scope of this project.

closed 2021-05-24 by anujpathania