anujpathania/HotSniper #13: error while loading shared libraries: libstlport-dynamic.so

cthi001, 2021-07-14:
I receive this error when trying to _make run_ a test case in _test_ folder.

I have no name!@fb1bb023ade4:/home/cthi4086/HotSniper/test/api$ make run
../../run-sniper -n 2 -c gainestown --roi -g --api/simple/cores_per_socket=1 -- ./api
[SNIPER] Start
/home/cthi4086/HotSniper/pin_kit/intel64/bin/pinbin: error while loading shared libraries: libstlport-dynamic.so: cannot open shared object file: No such file or directory
[SNIPER] End
[SNIPER] Elapsed time: 0.03 seconds
Makefile:8: recipe for target 'run_api' failed
make: *** [run_api] Error 127

anujpathania, 2021-07-15:
Hi. Please run benchmarks (PARSEC,SPLASH, and SPEC) through SimulationControl Framework; Step 4 in readme. Do you get the same error?

anujpathania, 2021-08-16:
I am closing this issue due to lack of response. Feel free to reopen the issue if the problem has not resolved.

closed 2021-08-16 by anujpathania