anujpathania/HotSniper #21: parsec.cs.princeton.edu's certificate expired

imsridhar, 2021-11-11:
ERROR: cannot verify parsec.cs.princeton.edu's certificate, issued by 'CN=R3,O=Let\'s Encrypt,C=US':
Issued certificate has expired
[image]

Working after adding '--no-check-certificate' in /benchmark/parsec/Makefile
line 6:	wget -O - http://parsec.cs.princeton.edu/download/2.1/parsec-2.1-core.tar.gz --no-check-certificate | tar xz --keep-old-files ; true
line21:	wget -O - http://parsec.cs.princeton.edu/download/2.1/parsec-2.1-sim.tar.gz --no-check-certificate | tar xz
line 25:	wget -O - http://parsec.cs.princeton.edu/download/2.1/parsec-2.1-native.tar.gz --no-check-certificate  | tar xz

anujpathania, 2021-11-11:
Hi Sridhar,

So this no longer an issue, since you have yourself made a fix?

imsridhar, 2021-11-11:
> Hi Sridhar,
> 
> So this no longer an issue, since you have yourself made a fix?

Yes, Sir. No issue right now. It would be good to make changes in the repository too.

anujpathania, 2021-11-11:
Can you please make changes and create a pull request in the repository.

imsridhar, 2021-11-11:
Sir, after making those changes following error comes up. This error is after the 'make' command in the 'Compiling the Benchmarks' section.

`
In file included from /usr/include/fcntl.h:289:0,
                 from /home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/parsec/parsec-2.1/./pkgs/libs/libxml2/src/nanohttp.c:49:
In function 'open',
    inlined from 'xmlNanoHTTPSave' at /home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/parsec/parsec-2.1/./pkgs/libs/libxml2/src/nanohttp.c:1585:12:
/usr/include/x86_64-linux-gnu/bits/fcntl2.h:50:4: error: call to '__open_missing_mode' declared with attribute error: open with O_CREAT or O_TMPFILE in second argument needs 3 arguments
    __open_missing_mode ();
    ^
make[4]: *** [nanohttp.lo] Error 1
Makefile:853: recipe for target 'nanohttp.lo' failed
make[4]: Leaving directory '/home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/parsec/parsec-2.1/pkgs/libs/libxml2/obj/amd64-linux.gcc-sniper'
make[3]: *** [all-recursive] Error 1
Makefile:1017: recipe for target 'all-recursive' failed
make[3]: Leaving directory '/home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/parsec/parsec-2.1/pkgs/libs/libxml2/obj/amd64-linux.gcc-sniper'
Makefile:560: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/parsec/parsec-2.1/pkgs/libs/libxml2/obj/amd64-linux.gcc-sniper'
[PARSEC] Error: 'env CFLAGS=-I/home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/parsec/parsec-2.1/pkgs/libs/zlib/inst/amd64-linux.gcc-sniper/include -O3 -funroll-loops -fprefetch-loop-arrays -static-libgcc -Wl,--hash-style=both,--as-needed -DPARSEC_VERSION=2.1 -DENABLE_PARSEC_HOOKS -I/home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/parsec/parsec-2.1/pkgs/libs/hooks/inst/amd64-linux.gcc-sniper/include -I/home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/tools/hooks -I/home/sridhar/Documents/csg/temp/HotSniper-master/include -DENABLE_PARMACS LDFLAGS=-L/home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/parsec/parsec-2.1/pkgs/libs/zlib/inst/amd64-linux.gcc-sniper/lib -L/usr/lib64 -L/usr/lib -L/home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/parsec/parsec-2.1/pkgs/libs/hooks/inst/amd64-linux.gcc-sniper/lib -uparmacs_roi_end -uparmacs_roi_start -L/home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/tools/hooks -lhooks_base -lrt -pthread /usr/bin/make' failed.
Makefile:31: recipe for target 'parsec-build' failed
make[1]: *** [parsec-build] Error 1
make[1]: Leaving directory '/home/sridhar/Documents/csg/temp/HotSniper-master/benchmarks/parsec'
Makefile:12: recipe for target 'all' failed
make: *** [all] Error 2

`

anujpathania, 2021-11-11:
Can you do a clean and build again from scratch.

closed 2022-02-07 by anujpathania