anujpathania/HotSniper #22: Not able to download pinplay-drdebug : Moved permanently

imsridhar, 2021-11-11:
[image]

anujpathania, 2021-11-11:
Hi Sridhar,

Unfortunately, we cannot distribute Pin directly through the repository according to the license from Intel. 

Do you know the new link where the Pin tool is moved?

imsridhar, 2021-11-11:
in the readme pinplay section, changing 'wget' to 'curl -O' works.
`wget https://software.intel.com/content/dam/develop/external/us/en/protected/pinplay-drdebug-3.2-pin-3.2-81205-gcc-linux.tar.gz
`
`curl -O https://software.intel.com/content/dam/develop/external/us/en/protected/pinplay-drdebug-3.2-pin-3.2-81205-gcc-linux.tar.gz
`

anujpathania, 2021-11-11:
It will be great if you can create a patch for both this issue and issues #21, when you have a stable fix ready. We will test the patch at our end independently, and then merge it into the main branch.

closed 2022-02-07 by anujpathania

anujpathania, 2022-05-22:
This updated command fixes the wget 403 forbidden error:

wget --user-agent="Mozilla"  https://www.intel.com/content/dam/develop/external/us/en/protected/pinplay-drdebug-3.2-pin-3.2-81205-gcc-linux.tar.gz