anujpathania/HotSniper #23: Possible solution to resolve spike in McPAT Power values?

gkothar1, 2022-02-07:
Hi,

During DVFS simulations in HotSniper, I have noticed that occasionally McPat will estimate an absurdly large area and power value for one or more processor functional blocks, which in turn results in a high-temperature value during HotSpot simulation. This produces incorrect results.

I suspect the reason for this is that McPat is invoked with option  `-opt_for_clk 1` (which is hardcoded to 1). This option optimizes the McPATs output based on target clock frequency.  

During DVFS simulation, the frequencies are bound to vary, and when McPat is invoked with a new clock frequency value, it generates an output optimized for the current frequency in which **the areas and peak power values** of the same functional block are quite different even if the microarchitectural specifications are exactly same between two successive invocations).

I wanted to verify from the authors whether indeed is this the reason and option  `-opt_for_clk` should be set to `0`. This optimizes the output for ED^2P (energy delay square product).

As an example for 4MB, 16-way NUCA (L3) Cache Bank at 22nm and 4.4 GHz target clock frequency, `McPat-1.3` reports:
- With `-opt_for_clk 1`: Area ~24 mm2
- With `-opt_for_clk 0`: Area ~8.4 mm2 (seems legit)

More info about `opt_for_clk`: [https://github.com/HewlettPackard/mcpat](https://github.com/HewlettPackard/mcpat)

Regards,
Gaurav

anujpathania, 2022-02-10:
Hi Gaurav,

We are looking into this, and get back to you.

-- Anuj

anujpathania, 2023-02-10:
Only slight changes were observed in the reported McPaT power based on "-opt_for_clk" power.

Nevertheless, agreed with your observation about Power/Area. Therefore, Default changed to "0" from "1" for -opt_for_clk.

closed 2023-02-10 by anujpathania