anujpathania/HotSniper #24: Voltage calculation from frequency in DVFS table

gkothar1, 2022-02-20:
Hi,

Referring to:
https://github.com/anujpathania/HotSniper/blob/a2c9193821c392b8860968efceeda2964245ffd5/scripts/energystats.py#L23

- What does each term in the formula to obtain a voltage for a given frequency denote? 
- Is this table valid when you have a max frequency greater than 4GHz?

Regards,
Gaurav

anujpathania, 2022-02-20:
Hi Gaurav,

The formula comes from an approximate fitting the voltage-frequency pairs of the original integration of Sniper and McPat, which looks like -

"if tech == 22:
    return [ (2000, 1.0), (1800, 0.9), (1500, 0.8), (1000, 0.7), (0, 0.6) ]"

The formula extends it to 4GHz and all other possible discrete DVFS levels between 0 and 4GHz. In code, the table starts from highest frequency to lowest frequency (zero). The highest frequency is set to 4GHz. 

You can potentially use this formula above 4 GHz also. However, it maybe useful to do a sanity check with a voltage-frequency pairs from real platform if you have access to such data. Thereafter, do some calibration to this formula accordingly. 

-- Anuj

closed 2022-02-24 by anujpathania

gkothar1, 2022-02-24:
Hi Anuj,

Thanks for the clarification.

Regards,
Gaurav