anujpathania/HotSniper #28: About NUCA Cache Setting

hisengberg, 2022-12-14:
Dear professors:
    I want to do some experiments with S-NUCA Cache and the config that i want is a many-core with L2 cache being the physically distributed LLC. (that means the cache levels is 2----the L1 cache(include i-cache and d-cache) is private for each core and L2 cache is shared by NoC).
    For this reason, I have made the following configuration changes:
    1.in `base.cfg`  `[perf_model/nuca/enabled = true]`, `[perf_model/l1_icache/perfect = true]`, `[perf_model/l1_dcache/perfect = true]`, `[perf_model/l2_cache/perfect = true]`
    2.in `nehalem.cfg` `[perf_model/cache/levels = 2]` and at the same time i annotate the informations about _l3_cahe_(i noticed that the default setting is three levels cache, and the l3_cache as the llc)
    But after these changes still can't succeed, can u tell me how should i set it up correctly?
    Hope to get your reply kindly.
    
Regards,
Walter

anujpathania, 2022-12-15:
Hi Walter,

I remember Sniper does not allow for NUCA in L2. 

You can set L2 as the same size as L1, and set the access latency for L2 as zero. This would make L2 non-existent (transparent). Then add NUCA as your physically-distributed logically-shared LLC with each core having an NUCA bank. This will make your NUCA a pseudo-L2 LLC.

-- Anuj

hisengberg, 2022-12-16:
Hi Anuj,

Thank you very much for your reply.

If I use this approach to set L2 LLC, then whether the `[perf_model/cache/levels]` in `nehalem.cfg` should set as 3 rather than 2?

and in `nuca-cache.cfg`:  when i specify:
[image]
then does this 2048KB is shared by all of the cores or each core will have 2048KB nuca-cache?

--Walter

dhriti1988, 2022-12-28:
> Hi Anuj,
>
> Thank you very much for your reply.
>
> If I use this approach to set L2 LLC, then whether the `[perf_model/cache/levels]` in `nehalem.cfg` should set as 3 rather than 2?
>
> and in [...]

Hi @hisengberg 
I am also trying similar setup where I need to enable nuca. Did you get any clarity on this ? please let me know.                   

cc: @anujpathania

hisengberg, 2022-12-28:
> 

Hi
i already known that the `perf_model/cache/levels ` is not include nuca cache, which means that if u set levels = 2, it means private L1-cache + L2 cache.
and the question 2 that i asked above, i still don't kown about that.

anujpathania, 2022-12-28:
Hi Walter and Dhriti,

Yes, for S-NUCA, you set perf_model/cache/levels=2. If you don't want L2 then make it transparent as I mentioned above.

**As per my understanding,** If you set cache_size = 2048, then your entire NUCA is of 2048 KB. So if you have 16 cores then each core has cache bank of 128KB.  Maybe double check it via the code.

hisengberg, 2022-12-28:
Hi Anuj,

Thank you for your reply, it helps us a lot.

closed 2022-12-28 by hisengberg