anujpathania/HotSniper #30: How to use tools/gen_topology.py

hisengberg, 2023-02-09:
Hi,
  I want to use `gen_topology.py` in `HotSniper/tools` directory to generate a topology file, and when i finished an experiment, i did the following steps but reported following error(in fact, this is the source code, and i haven't modified it).

  So, could you please let me know how to use `gen_topology.py` ?

Best wishes
[image]

anujpathania, 2023-02-09:
Hi,

gen_topology.py is a script inherited from original Sniper. The script is very finicky and works with only some architecture configurations.

There is nothing per-say I can think of in HotSniper that would break this script.

I recommend you verify that your architecture configuration is compatible with gen_topology.py in the original Sniper.

hisengberg, 2023-02-10:
Hi anuj,
  I've fixed this issue, because the script is made by python2 rather than python3.

closed 2023-02-10 by hisengberg