anujpathania/HotSniper #48: error: 'PIN_REGISTER' has not been declared & PinPlay 3.2 no longer available

primussvensson, 2024-01-11:
Hello!

I'm trying to run your simulation tool but run into issues with compiling it as I can not find PinPlay 3.2 anywhere. 
The link in the setup guide is no longer valid. I tried using the pinkit provided from the intel SDE 9 but compiling HotSniper results in the following error: emulation.cc:12:44: error: 'PIN_REGISTER' has not been declared
 static void handleRdtsc(THREADID threadid, PIN_REGISTER * gax, PIN_REGI
Any ideas for a fix would be greatly appreciated!

anujpathania, 2024-01-11:
Hello!

[Intel Simplified Software License](https://[software.intel.com/sites/landingpage/pintool/intel-simplified-software-license.txt](https://software.intel.com/sites/landingpage/pintool/intel-simplified-software-license.txt)) allows us to distribute Pin now.

I have committed PinPlay 3.2 to the repository. See the latest commits, and see if it solves the problem.

primussvensson, 2024-01-12:
Hello again!

Thank you for quick assistance, using the provided PinPlay solved the issue!
A side note, the link to parsec benchmark is no longer working, causing the make of the benchmarks to fail.

I just removed it from the make and run.py files and it works!

anujpathania, 2024-01-12:
Thanks. We know about the issue with PARSEC, and working towards fixing it.

closed 2024-01-12 by anujpathania