anujpathania/HotSniper #57: Clarification on task scheduling and task migration overheads

anuiiit, 2025-02-01:
Hi Sir,

We are working on a thread migration policy based on an existing initial thread mapping to balance power consumption and performance. 

We’re having trouble analyzing the runtime overhead of the thread migration policy and the scheduling overheads. Although the scheduler overheads can be captured as reschedule_cost in [perf_model/sync], the value is set by the user and cannot be computed in real time. It seems the scheduler runs on the host machine and not part of the simulation, E.G. when we ran an benchmark application with 16 threads, the periodic power consumption file (PeriodicPower.log.gz file) and execution logs showed only 16 threads running.

Please clarify few points.
1) If the scheduler runs on the simulated core or on the host machine?
2) If the scheduler indeed runs on the simulator core, how can the energy and power overheads be computed?
3) When we migrate a thread from one core to another core and the frequency of the destination core is different from the source core then both dvfs and migration penalities are considered or only one of the penality is considered?

anujpathania, 2025-02-01:
Hi,

1) Scheduler runs on the host machine.

2) Does not apply. There is no easy way to measure overhead of scheduler directly in live simulation. You can potentially compile the scheduler as binary and execute it on HotSniper with dummy input to estimate the overhead.

3) Their is no DVFS penalty. There is an inherent thread migration penalty, due to cold cache misses caused by migration.

closed 2025-02-04 by anujpathania

anuiiit, 2025-02-12:
Thanks a lot for clarification Sir!
I was travelling last week therefore I could not close the issue.